HuajianUP/Photo-SLAM #1: How can i run in ubuntn18.04LTS?

chengYi-xun, 2024-04-01:


chengYi-xun, 2024-04-02:
How can I fix this error? i use the ubuntn18.04LTS build it successfully,but i can not run it

chengYi-xun, 2024-04-02:
At the same time, when I run to here:
```    
std::filesystem::path pathImageDir(strImageDir);
pathImageDir /= "results";
```
it also throws an error“段错误 (核心已转储)
”. This seems to be an issue with `std::filesystem::path`, and I don't know how to resolve it.

liquorleaf, 2024-04-02:
Hi!
It looks like the default version of gcc/g++ provided in Ubuntu 18.04 LTS does not support std::filesystem, which is formally implemented in C++17. I'm sorry that we have not tested with this OS before. Still, we find some solutions that might work for you.[askubuntu.com](https://askubuntu.com/questions/1256440/how-to-get-libstdc-with-c17-filesystem-headers-on-ubuntu-18-bionic) says that there might be two possible solutions:
1. move from `<filesystem>` to `<experimental/filesystem>`
2. install gcc-9 and g++-9, then create soft links to let cmake use them.

chengYi-xun, 2024-04-03:
Thank you very much for your work. I have successfully run the code on Ubuntu 18.04. The error was resolved using the following method:
add to CMakeLists:
```
set(CMAKE_CXX_STANDARD_REQUIRED ON)
link_libraries(stdc++fs)
```

closed 2024-04-03 by chengYi-xun