HuajianUP/Photo-SLAM #2: I'm having some problems compiling your code.

Tianci-Wen, 2024-04-08:
Excellent work! I'm having some problems compiling, can you help me see if you've encountered any? Is there any solution?

my enviornment:
- OS	Ubuntu 20.04 LTS	
- gcc	9.4.0	
- cmake	3.22.1
- [CUDA]	11.8	
- [cuDNN]	8.9.3	
- [OpenCV]built with opencv_contrib and CUDA	4.7.0	
- [LibTorch](https://pytorch.org/get-started/locally)	cxx11-abi-shared-with-deps-2.0.1+cu118

`lzy@lzy:~/workingspace/Photo-SLAM$ ./build.sh 
mkdir: cannot create directory ‘build’: File exists
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/lzy/workingspace/Photo-SLAM/ORB-SLAM3/Thirdparty/DBoW2/build
Consolidate compiler generated dependencies of target DBoW2
[100%] Built target DBoW2
mkdir: cannot create directory ‘build’: File exists
CMake Deprecation Warning at CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lzy/workingspace/Photo-SLAM/ORB-SLAM3/Thirdparty/g2o/build
Consolidate compiler generated dependencies of target g2o
[100%] Built target g2o
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lzy/workingspace/Photo-SLAM/ORB-SLAM3/Thirdparty/Sophus/build
Consolidate compiler generated dependencies of target test_se2
Consolidate compiler generated dependencies of target test_so3
Consolidate compiler generated dependencies of target test_common
Consolidate compiler generated dependencies of target test_se3
Consolidate compiler generated dependencies of target test_so2
Consolidate compiler generated dependencies of target test_sim2
Consolidate compiler generated dependencies of target test_rxso3
Consolidate compiler generated dependencies of target test_rxso2
Consolidate compiler generated dependencies of target test_sim3
Consolidate compiler generated dependencies of target HelloSO3
Consolidate compiler generated dependencies of target test_geometry
Consolidate compiler generated dependencies of target test_velocities
[  8%] Built target test_se2
[ 16%] Built target test_se3
[ 25%] Built target test_so3
[ 33%] Built target test_so2
[ 41%] Built target test_common
[ 50%] Built target test_sim2
[ 58%] Built target test_sim3
[ 66%] Built target test_rxso2
[ 75%] Built target test_geometry
[ 83%] Built target test_velocities
[ 91%] Built target test_rxso3
[100%] Built target HelloSO3
Uncompress vocabulary ...
mkdir: cannot create directory ‘build’: File exists
[... 238 lines cut ...]
/usr/bin/ld: replica_rgbd.cpp:(.text.startup+0x164a): undefined reference to `ORB_SLAM3::System::SaveTrajectoryKITTI(std::string const&)'
/usr/bin/ld: ../lib/libgaussian_viewer.so: undefined reference to `cv::FileStorage::FileStorage(std::string const&, int, std::string const&)'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `Json::Value::Value(std::string const&)'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `cv::FileNode::string() const'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/replica_rgbd.dir/build.make:186: ../bin/replica_rgbd] Error 1
make[1]: *** [CMakeFiles/Makefile2:333: CMakeFiles/replica_rgbd.dir/all] Error 2
/usr/bin/ld: CMakeFiles/train_colmap.dir/examples/train_colmap.cpp.o: in function `readColmapImagesBinary(std::shared_ptr<GaussianMapper>, std::filesystem::path const&, std::filesystem::path const&)':
train_colmap.cpp:(.text+0xc39): undefined reference to `cv::imread(std::string const&, int)'
/usr/bin/ld: CMakeFiles/train_colmap.dir/examples/train_colmap.cpp.o: in function `readColmapCamerasBinary(std::shared_ptr<GaussianMapper>, std::filesystem::path const&)':
train_colmap.cpp:(.text+0x351f): undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: train_colmap.cpp:(.text+0x356d): undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: train_colmap.cpp:(.text+0x35a4): undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: train_colmap.cpp:(.text+0x35db): undefined reference to `cv::error(int, std::string const&, char const*, char const*, int)'
/usr/bin/ld: ../lib/libgaussian_viewer.so: undefined reference to `cv::FileStorage::FileStorage(std::string const&, int, std::string const&)'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `Json::Value::Value(std::string const&)'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `cv::FileNode::string() const'
/usr/bin/ld: ../lib/libgaussian_mapper.so: undefined reference to `cv::imwrite(std::string const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/train_colmap.dir/build.make:186: ../bin/train_colmap] Error 1
make[1]: *** [CMakeFiles/Makefile2:243: CMakeFiles/train_colmap.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
`

liquorleaf, 2024-04-09:
Seems like your linker failed to find symbols of opencv, jsoncpp, ORB_SLAM3, etc. Have you tried a newer version of cmake? Did you change anything in CMakeLists.txt?

Tianci-Wen, 2024-04-09:
> Seems like your linker failed to find symbols of opencv, jsoncpp, ORB_SLAM3, etc. Have you tried a newer version of cmake? Did you change anything in CMakeLists.txt?

I update my cmake to 3.27.5, but it resport the same error that it failed to link opencv, ORB_SLAM3, etc. 

For the CMakeLists.txt, I just add `set(Torch_DIR /home/lzy/workingspace/Photo-SLAM/lib/libtorch/share/cmake/Torch)`. Now I'm trying to set path for these library.

For the build.sh, I modify `make -j8` to  make `-j128 `. I have tried with `make -j8`, same error.
```
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release # add OpenCV_DIR definitions if needed, example:
#cmake .. -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/home/rapidlab/libs/opencv/lib/cmake/opencv4
make -j128

# Photo-SLAM
echo "Building Photo-SLAM ..."
cd ../..
mkdir build
cd build
cmake .. # add Torch_DIR and/or OpenCV_DIR definitions if needed, example:
#cmake .. -DTorch_DIR=/home/rapidlab/libs/libtorch/share/cmake/Torch -DOpenCV_DIR=/home/rapidlab/libs/opencv/lib/cmake/opencv4
make -j128
```

Here is my CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.20 FATAL_ERROR)
project(photo-slam LANGUAGES CXX CUDA)

if(NOT CMAKE_BUILD_TYPE)
    # set(CMAKE_BUILD_TYPE Debug)
    set(CMAKE_BUILD_TYPE Release)
endif()

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_CUDA_STANDARD 17)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wno-deprecated-declarations")

set(CAFFE2_USE_CUDNN on)

# Uncomment the following line if you want to solve CMake Warning "Failed to compute shorthash for libnvrtc.so"
# set(PYTHON_EXECUTABLE "/usr/bin/python3")
set(Torch_DIR /home/lzy/workingspace/Photo-SLAM/lib/libtorch/share/cmake/Torch)

find_package(Torch REQUIRED)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${TORCH_CXX_FLAGS} -pthread")

# set(ENV{CUDA_LAUNCH_BLOCKING} 1)
# set(ENV{TORCH_USE_CUDA_DSA} 1)

find_package(CUDA REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(OpenCV 4 REQUIRED)
find_package(jsoncpp REQUIRED)

find_package(OpenGL REQUIRED)
find_package(glm REQUIRED)
find_package(glfw3 REQUIRED)

find_package(realsense2)

set(ORB_SLAM3_SOURCE_DIR "${PROJECT_SOURCE_DIR}/ORB-SLAM3")
set(CMAKE_LIBRARY_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/lib)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${PROJECT_SOURCE_DIR}/bin)

include_directories(
    ${PROJECT_SOURCE_DIR}
    ${ORB_SLAM3_SOURCE_DIR}
[... 176 lines cut ...]
    ${ORB_SLAM3_SOURCE_DIR}/lib/libORB_SLAM3.so)

# EuRoC Stereo
add_executable(euroc_stereo examples/euroc_stereo.cpp)
target_link_libraries(euroc_stereo
    gaussian_viewer    
    gaussian_mapper
    ${ORB_SLAM3_SOURCE_DIR}/lib/libORB_SLAM3.so)

##################################################################################
##  Build the mapping examples to ${PROJECT_SOURCE_DIR}/bin
##################################################################################

# Intel Realsense
if(realsense2_FOUND)
    add_executable(realsense_rgbd examples/realsense_rgbd.cpp)
    target_include_directories(realsense_rgbd PUBLIC
        ${realsense_INCLUDE_DIR})
    target_link_libraries(realsense_rgbd
        gaussian_viewer    
        gaussian_mapper
        ${ORB_SLAM3_SOURCE_DIR}/lib/libORB_SLAM3.so
        ${realsense2_LIBRARY})
endif()
```

Tianci-Wen, 2024-04-09:
I would like to ask if you can provide a docker container, so that there should be no problem with the environment.

liquorleaf, 2024-04-09:
Could you find the target symbols in the .so libs by command like
```
nm -C ./ORB-SLAM3/lib/libORB_SLAM3.so | grep "TrackMonocular"
```

Tianci-Wen, 2024-04-09:
> Could you find the target symbols in the .so libs by command like ?
> 
> ```
> nm -C ./ORB-SLAM3/lib/libORB_SLAM3.so | grep "TrackMonocular"
> ```

It does seem to find this function in libORB_SLAM3.so :
[image]

Tianci-Wen, 2024-04-09:
> It does seem to find this function in libORB_SLAM3.so : [...]

What it want is `ORB_SLAM3::System::TrackMonocular(cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::string)'`. In libORB_SLAM3.so, it is  
```
lzy@lzy:~/workingspace/Photo-SLAM$ nm -C ./ORB-SLAM3/lib/libORB_SLAM3.so | grep "ORB_SLAM3::System::TrackMonocular"
0000000000142bc0 T ORB_SLAM3::System::TrackMonocular(cv::Mat const&, double const&, std::vector<ORB_SLAM3::IMU::Point, std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >).
```

They are different in the final input parameter `std::string` and `std::allocator<ORB_SLAM3::IMU::Point> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >`.  

I'm wondering if there's something wrong with my gcc version.

closed 2024-04-11 by Tianci-Wen

fengDd11, 2024-04-17:
同学您好，我也遇到了相同的问题，我也找不出原因是什么

mumu2000mumu, 2024-04-17:
同学您好，我也遇到了相同的问题，并且找不出原因，

Tianci-Wen, 2024-04-17:
> 同学您好，我也遇到了相同的问题，并且找不出原因，

我建议使用docker配置你的环境，并且把所有环境与这个代码中要求配置一致

mumu2000mumu, 2024-04-17:
好的谢谢

mumu2000mumu, 2024-04-17:
> > 同学您好，我也遇到了相同的问题，并且找不出原因，
> 
> 我建议使用docker配置你的环境，并且把所有环境与这个代码中要求配置一致
同学您好，我还是解决不了该问题，可以麻烦同学分享一下docker吗，十分感谢同学的指导帮助。

Eragonlrc, 2024-04-20:
同学您好，我也在环境配置上遇到了问题，请问您是否方便分享一下使用的docker镜像？

XuYinzhe, 2024-05-02:
> > 同学您好，我也遇到了相同的问题，并且找不出原因，
> 
> 我建议使用docker配置你的环境，并且把所有环境与这个代码中要求配置一致

您好请问能分享一下您的dockerfile吗？我也是照着要求配置，但是还是编译失败了。

这个是我的docker:
`
FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
	wget https://download.pytorch.org/libtorch/cu118/libtorch-cxx11-abi-shared-with-deps-2.0.1%2Bcu118.zip -O libtorch-cu118.zip && \
	unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM
`

XuYinzhe, 2024-05-02:
I can successfully compile using the Dockerfile:

```
FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    libssl-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
	wget https://download.pytorch.org/libtorch/cu118/libtorch-cxx11-abi-shared-with-deps-2.0.1%2Bcu118.zip -O libtorch-cu118.zip && \
	unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM

```

fealty94, 2024-05-13:
> I can successfully compile using the Dockerfile:
>
> ```
> FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04
>
> ENV DEBIAN_FRONTEND=noninteractive
>
> RUN apt-get update && apt-get install -y \ [...]

This works for me. Thank you!
But now, imgui_viewer makes Segmentation fault.
How did you fix it?

vulnic, 2024-06-07:
[image]
Had an issue with the build.sh using the above dockerfile, but solved it.

Adding Torch_DIR to the last `cmake ..` call (as suggested in the comment) of the `build.sh` solved it for me! 
I did not add the `-DOpen_CV` variable, as I could not find the equivalent directory within the container.
- Changed to `cmake .. -DTorch_DIR=/libtorch/share/cmake/Torch` without  `-DOpen_CV`

buzhichunfeng, 2024-07-20:
你好，这个项目可以生成稠密点云和轨迹图吗？

buzhichunfeng, 2024-07-20:
同时，这个项目如何部署在真实世界中。并且在docker中没有图形化节面改怎么查看结果呢？

Looong01, 2024-11-19:
> 同时，这个项目如何部署在真实世界中。并且在docker中没有图形化节面改怎么查看结果呢？

可以将画面映射到宿主机

showmaker00, 2024-11-26:
I had the same issue and ended up finding out that for me it was the wrong version of libtorch, mine was libtorch-shared-with-deps-2.0.1+cu118, which had to be replaced with libtorch-cxx11-abi-shared-with-deps-2.0.1+cu118 mentioned in this project.

mirpouya, 2025-01-16:
> I can successfully compile using the Dockerfile:
>
> ```
> FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04
>
> ENV DEBIAN_FRONTEND=noninteractive
>
> RUN apt-get update && apt-get install -y \ [...]

there's good guideline in #5 
I solved the issue with this docker file:

FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    libssl-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
    wget https://download.pytorch.org/libtorch/nightly/cu118/libtorch-cxx11-abi-shared-with-deps-2.4.0.dev20240425%2Bcu118.zip -O libtorch-cu118.zip && \
    unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM

everything is the same as https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703
but the libtorch version is 2.4.0

then you must modify all of:
     c10::guts::to_string(param.unsafeGetTensorImpl());
to:
     param.unsafeGetTensorImpl();
which is mentioned in #5 

and you will see the visualization