HuajianUP/Photo-SLAM #4: Errors about Installation "libhdf5_serial.so.103: undefined reference to curl" in OpenCV Make

shuttworth, 2024-04-13:
Thank you for your work. Some error happened when MAKE Opencv 4.8.0 in Ubuntu 22.04 as the code saying.
`[ 37%] Linking CXX executable ../../bin/opencv_test_hdf
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_global_init@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_perform@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_slist_free_all@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_setopt@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_init@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_slist_append@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libhdf5_serial.so.103: undefined reference to `curl_global_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
make[2]: *** [modules/hdf/CMakeFiles/opencv_test_hdf.dir/build.make:123: bin/opencv_test_hdf] Error 1
make[1]: *** [CMakeFiles/Makefile2:4664: modules/hdf/CMakeFiles/opencv_test_hdf.dir/all] Error 2`
How can i solve it ? Thank you

liquorleaf, 2024-04-13:
Seems like your linker cannot find symbols in CURL. Have you tried `sudo apt install curl`, or linking -lcurl manually?

shuttworth, 2024-04-14:
> Seems like your linker cannot find the CURL symbols. Have you tried `sudo apt install curl libcurl4-openssl-dev`, or linking -lcurl manually?

Thanks for your reply. I did some try aboyt `sudo apt install libcurl4-openssl-dev`,but not `sudo apt install curl libcurl4-openssl-dev`, your advice worked!  Problems Solved

closed 2024-04-14 by shuttworth