HuajianUP/Photo-SLAM #5: An Error in imgui_viewer

wangwang-xyz, 2024-04-15:
There is a problem when I run `./bin/tum_mono` or `./bin/gaussian_viewer`. The Error like this
```shell
Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:
Segmentation fault
```
Code related with this problem in `imgui_viewer.cpp` line 363
```cpp
glClearColor(0.0f, 0.0f, 0.0f, 1.0f);
        // Draw main window image
        if (show_main_rendered_)
        {
            auto drawlist = ImGui::GetBackgroundDrawList();
            if (pSLAM_ && tracking_vision_)
            {
                drawlist->AddImage((void *)(intptr_t)rendered_img_texture, ImVec2(0, 0),
                                   ImVec2(glfw_window_width_, glfw_window_height_));
            }
            else
            {
                cv::Mat main_img = pGausMapper_->renderFromPose(
                    Tcw_main_, rendered_image_width_main_, rendered_image_height_main_, true);
                cv::Mat main_img_to_show = cv::Mat(rendered_image_height_main_, padded_main_image_width_, CV_32FC3, cv::Vec3f(0.0f, 0.0f, 0.0f));
                main_img.copyTo(main_img_to_show(image_rect_main));
                glBindTexture(GL_TEXTURE_2D, main_img_texture);
                glTexImage2D(GL_TEXTURE_2D, 0, GL_RGB, main_img_to_show.cols, main_img_to_show.rows,
                     0, GL_RGB, GL_FLOAT, (float*)main_img_to_show.data);
                drawlist->AddImage((void *)(intptr_t)main_img_texture, ImVec2(0, 0),
                                   ImVec2(glfw_window_width_, glfw_window_height_));
            }
        }
```
However when I write a `test.cpp` to test this code, it works well. The `test.cpp` like this:
```cpp
/* Copyright 2021 iwatake2222

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at

    http://www.apache.org/licenses/LICENSE-2.0

Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
==============================================================================*/
/*** Include ***/
/* for general */
#include <cstdint>
#include <cstdio>
#include <fstream> 
#include <vector>
#include <string>
#include <chrono>

/* for GLFW */
#include <GLFW/glfw3.h>

/* for ImGui */
#include "viewer/imgui/imgui.h"
#include "viewer/imgui/imgui_impl_glfw.h"
#include "viewer/imgui/imgui_impl_opengl3.h"

#include <glm/glm.hpp>
#include <glm/gtc/matrix_transform.hpp>
#include <glm/gtc/type_ptr.hpp>
#include <glm/gtx/string_cast.hpp>
[... 172 lines cut ...]
    /*** Finalize ***/
    /* imgui: cleanup */
    ImGui_ImplOpenGL3_Shutdown();
    ImGui_ImplGlfw_Shutdown();
    ImGui::DestroyContext();

    /* Close OpenGL window and terminate GLFW */
    glfwDestroyWindow(window);
    glfwTerminate();

    return 0;
}
```
The related `CMakeLists.txt` like this:
```
add_executable(test examples/test.cpp)
target_link_libraries(test
    # gaussian_viewer
    glm::glm
    glfw
    OpenGL::GL
    imgui
    ${OpenCV_LIBRARIES}
)
```

liquorleaf, 2024-04-15:
Hi!
Which version of CUDA and LibTorch did you use?

wangwang-xyz, 2024-04-16:
The environment I used is：
* cuda 11.8
* LibTorch 2.0.1
* WSL2

It seems this problem caused by LibTorch which built with llvm.
I flowed this issue to update LibTorch to nightly version(2.4.0.dev20240415+cu118) and solved the problem
pytorch/pytorch#103756

In addition, LibTorch removed `c10::guts::to_string` in nightly version. Therefore, `gaussian_model.cpp` need to be modified.
Replace  all
```cpp
key = c10::guts::to_string(param.UnsafeGetTensorImpl());
```
 with 
```cpp
key = param.UnsafeGetTensorImpl();
```

Eragonlrc, 2024-04-26:
Similar segmentation fault error solved thanks to this issue!
BYW in LibTorch nightly version (2.4.0.dev20240425+cu118) replace all
```
key = c10::guts::to_string(param.UnsafeGetTensorImpl());
```
with
```
key = param.unsafeGetTensorImpl();
```
which uses camel case.

zhenzhongcao, 2024-05-10:
> There is a problem when I run `./bin/tum_mono` or `./bin/gaussian_viewer`. The Error like this
> 
> ```shell
> Pass 'Combine redundant instructions' is not initialized.
> Verify if there is a pass dependency cycle.
> Required Passes:
> Segmentation fault
> ```
> 
> Code related with this problem in `imgui_viewer.cpp` line 363
> 
> ```c++
> glClearColor(0.0f, 0.0f, 0.0f, 1.0f);
>         // Draw main window image
>         if (show_main_rendered_)
>         {
>             auto drawlist = ImGui::GetBackgroundDrawList();
>             if (pSLAM_ && tracking_vision_)
>             {
>                 drawlist->AddImage((void *)(intptr_t)rendered_img_texture, ImVec2(0, 0),
>                                    ImVec2(glfw_window_width_, glfw_window_height_));
>             }
>             else
>             {
>                 cv::Mat main_img = pGausMapper_->renderFromPose(
>                     Tcw_main_, rendered_image_width_main_, rendered_image_height_main_, true);
>                 cv::Mat main_img_to_show = cv::Mat(rendered_image_height_main_, padded_main_image_width_, CV_32FC3, cv::Vec3f(0.0f, 0.0f, 0.0f));
>                 main_img.copyTo(main_img_to_show(image_rect_main));
>                 glBindTexture(GL_TEXTURE_2D, main_img_texture);
>                 glTexImage2D(GL_TEXTURE_2D, 0, GL_RGB, main_img_to_show.cols, main_img_to_show.rows,
>                      0, GL_RGB, GL_FLOAT, (float*)main_img_to_show.data);
>                 drawlist->AddImage((void *)(intptr_t)main_img_texture, ImVec2(0, 0),
>                                    ImVec2(glfw_window_width_, glfw_window_height_));
>             }
>         }
> ```
> 
> However when I write a `test.cpp` to test this code, it works well. The `test.cpp` like this:
> 
> ```c++
> /* Copyright 2021 iwatake2222
> 
> Licensed under the Apache License, Version 2.0 (the "License");
> you may not use this file except in compliance with the License.
> You may obtain a copy of the License at
> 
>     http://www.apache.org/licenses/LICENSE-2.0
> 
> Unless required by applicable law or agreed to in writing, software
> distributed under the License is distributed on an "AS IS" BASIS,
> WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
> See the License for the specific language governing permissions and
> limitations under the License.
> ==============================================================================*/
> /*** Include ***/
> /* for general */
> #include <cstdint>
> #include <cstdio>
> #include <fstream> 
> #include <vector>
> #include <string>
> #include <chrono>
> 
> /* for GLFW */
> #include <GLFW/glfw3.h>
> 
> /* for ImGui */
> #include "viewer/imgui/imgui.h"
> #include "viewer/imgui/imgui_impl_glfw.h"
> #include "viewer/imgui/imgui_impl_opengl3.h"
[... 185 lines cut ...]
> 
>     /* Close OpenGL window and terminate GLFW */
>     glfwDestroyWindow(window);
>     glfwTerminate();
> 
>     return 0;
> }
> ```
> 
> The related `CMakeLists.txt` like this:
> 
> ```
> add_executable(test examples/test.cpp)
> target_link_libraries(test
>     # gaussian_viewer
>     glm::glm
>     glfw
>     OpenGL::GL
>     imgui
>     ${OpenCV_LIBRARIES}
> )
> ```

Did you run this code successfully? I encountered this problem. Can you help me answer it? Thank you.
[image]

alexjunholee, 2024-06-03:
In my case, the error was showing the exact same error message.

it was just a segfault error like below:
``` 104 Segmentation fault      (core dumped) ```

However, changing to nightly build as recommended from @wangwang-xyz and @Eragonlrc solved the issue! Thanks! 

*Note that you need to remove the whole build folder when re-building with the new libtorch.

Jiyu-Yao, 2024-06-10:
Thank you for your solution!
However,I met some trouble after I update LibTorch to nightly version.The error like this
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
/usr/bin/ld: /Photo-SLAM/lib/libgaussian_mapper.so: undefined reference to `Json::Value::Value(std::string const&)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::putText(cv::_InputOutputArray const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to `cv::FileNode::string[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/view_result.dir/build.make:184: /Photo-SLAM/bin/view_result] Error 1
make[1]: *** [CMakeFiles/Makefile2:273: CMakeFiles/view_result.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

It seems the opencv mismatch with the libtorch?I have no idea about this. Have you met this trouble?Could you please teach me how to solve it? Thank you!

> The environment I used is：
> 
>     * cuda 11.8
> 
>     * LibTorch 2.0.1
> 
>     * WSL2
> 
> 
> It seems this problem caused by LibTorch which built with llvm. I flowed this issue to update LibTorch to nightly version(2.4.0.dev20240415+cu118) and solved the problem [pytorch/pytorch#103756](https://github.com/pytorch/pytorch/issues/103756)
> 
> In addition, LibTorch removed `c10::guts::to_string` in nightly version. Therefore, `gaussian_model.cpp` need to be modified. Replace all
> 
> ```c++
> key = c10::guts::to_string(param.UnsafeGetTensorImpl());
> ```
> 
> with
> 
> ```c++
> key = param.UnsafeGetTensorImpl();
> ```

> The environment I used is：
> 
>     * cuda 11.8
> 
>     * LibTorch 2.0.1
> 
>     * WSL2
> 
> 
> It seems this problem caused by LibTorch which built with llvm. I flowed this issue to update LibTorch to nightly version(2.4.0.dev20240415+cu118) and solved the problem [pytorch/pytorch#103756](https://github.com/pytorch/pytorch/issues/103756)
> 
> In addition, LibTorch removed `c10::guts::to_string` in nightly version. Therefore, `gaussian_model.cpp` need to be modified. Replace all
> 
> ```c++
> key = c10::guts::to_string(param.UnsafeGetTensorImpl());
> ```
> 
> with
> 
> ```c++
> key = param.UnsafeGetTensorImpl();
> ```

> The environment I used is：
> 
>     * cuda 11.8
> 
>     * LibTorch 2.0.1
> 
>     * WSL2
> 
> 
> It seems this problem caused by LibTorch which built with llvm. I flowed this issue to update LibTorch to nightly version(2.4.0.dev20240415+cu118) and solved the problem [pytorch/pytorch#103756](https://github.com/pytorch/pytorch/issues/103756)
> 
> In addition, LibTorch removed `c10::guts::to_string` in nightly version. Therefore, `gaussian_model.cpp` need to be modified. Replace all
> 
> ```c++
> key = c10::guts::to_string(param.UnsafeGetTensorImpl());
> ```
> 
> with
> 
> ```c++
> key = param.UnsafeGetTensorImpl();
> ```

Jiyu-Yao, 2024-06-11:
Thank you for your solution!
However,I met some trouble after I update LibTorch to nightly version.The error like this
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)' /usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::operator<<(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)' /usr/bin/ld: /Photo-SLAM/lib/libgaussian_mapper.so: undefined reference to Json::Value::Value(std::string const&)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)' /usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::putText(cv::_InputOutputArray const&, std::cxx11::basic_string<char, std::char_traits, std::allocator > const&, cv::Point, int, double, cv::Scalar, int, int, bool)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::FileStorage::operator[](std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const' /usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&, int)'
/usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::write(cv::FileStorage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double)' /usr/bin/ld: /Photo-SLAM/ORB-SLAM3/lib/libORB_SLAM3.so: undefined reference to cv::FileNode::string[abi:cxx11](https://github.com/HuajianUP/Photo-SLAM/issues/5) const'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/view_result.dir/build.make:184: /Photo-SLAM/bin/view_result] Error 1
make[1]: *** [CMakeFiles/Makefile2:273: CMakeFiles/view_result.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

It seems the opencv mismatch with the libtorch?I have no idea about this. Have you met this trouble?Could you please teach me how to solve it? Thank you!

HuajianUP, 2024-06-11:
Hi Jiyu,

I reproduced this error when using the latest version of LibTorch, regardless of the OpenCV version. We will investigate further and try to support the latest version of LibTorch in the future.

Please use the tested LibTorch version first and then you should be able to compile the code successfully.

mirpouya, 2025-01-16:
for me:

there's good guideline in #5 
I solved the issue with this docker file:

FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    libssl-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
    wget https://download.pytorch.org/libtorch/nightly/cu118/libtorch-cxx11-abi-shared-with-deps-2.4.0.dev20240425%2Bcu118.zip -O libtorch-cu118.zip && \
    unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM

everything is the same as https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703
but the libtorch version is 2.4.0

then you must modify all of:
     c10::guts::to_string(param.unsafeGetTensorImpl());
to:
     param.unsafeGetTensorImpl();
which is mentioned in #5 

and you will see the visualization