HuajianUP/Photo-SLAM #8: About details of  Gaussian pyramid

lucky9-cyou, 2024-05-09:
Thanks for your wonderful code and paper.

I would like to ask some details about the Gaussian pyramid. As training progresses, it is mentioned in the paper that Gaussian smoothing and downsampling operations will be continuously used to obtain a lower Gaussian pyramid. I would like to ask, during this process, whether the previous Gaussian pyramid will continue to train and whether it will downsample the incoming pictures at the same time. And after training, how to perform inference using multiple Gaussian pyramids.

zhenzhongcao, 2024-05-10:
> Thanks for your wonderful code and paper.
>
> I would like to ask some details about the Gaussian pyramid. As training progresses, it is mentioned in the paper that Gaussian smoothing and downsampling [...]

Did you run this code successfully? I encountered this problem. Can you help me answer it? Thank you.
[image]

liquorleaf, 2024-05-23:
In our paper and implementation,  GausPyramid.do means each Keyframe is downsampled when used for the first few times [(GausPyramid.sub_level_times_of_use per level) * GausPyramid.num_sub_levels] after it enter GaussianMapper. The pyramid of each Keyframe is independent from that of one another. And each iteration only takes one Keyframe into account and consumes just one time of using. After all the GaussianPyramid using times exhausts, it will join the iteration who chooses it in primary resolution.

lucky9-cyou, 2024-05-29:
> In our paper and implementation, GausPyramid.do means each Keyframe is downsampled when used for the first few times [(GausPyramid.sub_level_times_of_use per level) * GausPyramid.num_sub_levels] [...]

Thanks for your answer, I will check the code again.

closed 2024-06-03 by lucky9-cyou