HuajianUP/Photo-SLAM #12: An error occured while compiling your code.

Jiyu-Yao, 2024-06-08:
Thank you for your excellent work! 
When I try to compile your code,I get some trouble.Could you please check for the error?If you have encountered it,please tell me the solution.Thank you!

CMake Warning at CMakeLists.txt:36 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/yjy/Photo-SLAM/build
Consolidate compiler generated dependencies of target imgui
Consolidate compiler generated dependencies of target simple_knn
[ 17%] Built target imgui
Consolidate compiler generated dependencies of target cuda_rasterizer
[ 24%] Built target simple_knn
[ 40%] Built target cuda_rasterizer
Consolidate compiler generated dependencies of target gaussian_mapper
[ 42%] Building CXX object CMakeFiles/gaussian_mapper.dir/src/gaussian_model.cpp.o
/home/yjy/Photo-SLAM/src/gaussian_model.cpp: In member function ‘at::Tensor GaussianModel::replaceTensorToOptimizer(at::Tensor&, int)’:
/home/yjy/Photo-SLAM/src/gaussian_model.cpp:571:27: error: ‘to_string’ is not a member of ‘c10::guts’
  571 |     auto key = c10::guts::to_string(param.unsafeGetTensorImpl());
      |                           ^~~~~~~~~
/home/yjy/Photo-SLAM/src/gaussian_model.cpp:571:27: note: suggested alternatives:
In file included from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
                 from /usr/include/c++/9/memory:80,
                 from /home/yjy/Photo-SLAM/include/gaussian_model.h:18,
                 from /home/yjy/Photo-SLAM/src/gaussian_model.cpp:16:
/usr/include/c++/9/bits/basic_string.h:6601:3: note:   ‘std::to_string’
 6601 |   to_string(long double __val)
      |   ^~~~~~~~~
/usr/include/c++/9/bits/basic_string.h:6549:3: note:   ‘std::to_string’
 6549 |   to_string(int __val)
      |   ^~~~~~~~~
In file included from /home/yjy/libtorch/include/ATen/ThreadLocalState.h:14,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/graph_task.h:2,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/function.h:6,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/custom_function.h:7,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/autograd.h:5,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/all.h:7,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/torch.h:3,
                 from /home/yjy/Photo-SLAM/include/gaussian_model.h:24,
                 from /home/yjy/Photo-SLAM/src/gaussian_model.cpp:16:
/home/yjy/libtorch/include/c10/core/impl/TorchDispatchModeTLS.h:61:21: note:   ‘c10::impl::to_string’
   61 | C10_API std::string to_string(TorchDispatchModeKey mode_key);
      |                     ^~~~~~~~~
/home/yjy/Photo-SLAM/src/gaussian_model.cpp:581:22: error: ‘to_string’ is not a member of ‘c10::guts’
  581 |     key = c10::guts::to_string(param.unsafeGetTensorImpl());
      |                      ^~~~~~~~~
/home/yjy/Photo-SLAM/src/gaussian_model.cpp:581:22: note: suggested alternatives:
In file included from /usr/include/c++/9/string:55,
                 from /usr/include/c++/9/stdexcept:39,
                 from /usr/include/c++/9/array:39,
                 from /usr/include/c++/9/tuple:39,
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
[... 173 lines cut ...]
                 from /usr/include/c++/9/bits/unique_ptr.h:37,
                 from /usr/include/c++/9/memory:80,
                 from /home/yjy/Photo-SLAM/include/gaussian_model.h:18,
                 from /home/yjy/Photo-SLAM/src/gaussian_model.cpp:16:
/usr/include/c++/9/bits/basic_string.h:6601:3: note:   ‘std::to_string’
 6601 |   to_string(long double __val)
      |   ^~~~~~~~~
/usr/include/c++/9/bits/basic_string.h:6549:3: note:   ‘std::to_string’
 6549 |   to_string(int __val)
      |   ^~~~~~~~~
In file included from /home/yjy/libtorch/include/ATen/ThreadLocalState.h:14,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/graph_task.h:2,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/function.h:6,
                 from /home/yjy/libtorch/include/torch/csrc/autograd/custom_function.h:7,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/autograd.h:5,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/all.h:7,
                 from /home/yjy/libtorch/include/torch/csrc/api/include/torch/torch.h:3,
                 from /home/yjy/Photo-SLAM/include/gaussian_model.h:24,
                 from /home/yjy/Photo-SLAM/src/gaussian_model.cpp:16:
/home/yjy/libtorch/include/c10/core/impl/TorchDispatchModeTLS.h:61:21: note:   ‘c10::impl::to_string’
   61 | C10_API std::string to_string(TorchDispatchModeKey mode_key);
      |                     ^~~~~~~~~
make[2]: *** [CMakeFiles/gaussian_mapper.dir/build.make:104：CMakeFiles/gaussian_mapper.dir/src/gaussian_model.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:248：CMakeFiles/gaussian_mapper.dir/all] 错误 2
make: *** [Makefile:91：all] 错误 2

HuajianUP, 2024-06-11:
Hi Jiyu,

We notice that the system is not compatible with the latest Pytorch. Please use the tested version.
```
wget https://download.pytorch.org/libtorch/cu118/libtorch-cxx11-abi-shared-with-deps-2.0.1%2Bcu118.zip -O libtorch-cu118.zip
unzip libtorch-cu118.zip && rm libtorch-cu118.zip

# in CMakeLists.txt
set(Torch_DIR ./libtorch/share/cmake/Torch)
```

Jiyu-Yao, 2024-06-11:
> Hi Jiyu,
>
> We notice that the system is not compatible with the latest Pytorch. Please use the tested version.
>
> ```
> wget [...]

Thank you so much about your kind reply! 
I've noticed this issue and I compile your code successfully in docker with the dockerfile provided by https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703.
However, when I try to run it, it seems another error occurred. So confusing it is that seems nobody have met this before. Could you please check for it and provide some suggestion?
Thanks a lot!
[image]

xiangw369, 2024-06-14:
Please tell me how to solve this situation? Thanks!

Pass 'Combine redundant instructions' is not initialized.
Verify if there is a pass dependency cycle.
Required Passes:

HuajianUP, 2024-06-16:
Hi Xiang, Sorry, I don't get your questions. Please provide further details.

LeihanChen, 2024-08-09:
@HuajianUP @Jiyu-Yao Hi all, I encountered the same issue with the dockerfile provided by https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703,  hard to debug with the segmentation fault, any updates with this issue?
[image]
[image]

ZCWzy, 2024-08-24:
have same question, too

This is my debugging process

```
root@cf7685d5dec7:/Photo-SLAM# ./bin/replica_rgbd     ./ORB-SLAM3/Vocabulary/ORBvoc.txt     ./cfg/ORB_SLAM3/RGB-D/Replica/office0.yaml     ./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml     /data/download/Photo-SLAM-main/data/Replica/office0     /data/download/
CUDA available! Training on GPU.

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
Loading settings from ./cfg/ORB_SLAM3/RGB-D/Replica/office0.yaml
        -Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
        -Loaded image info
        -Loaded RGB-D calibration
        -Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
        -Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
        -Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
        -Loaded misc parameters
----------------------------------
SLAM settings: 
        -Camera 1 parameters (Pinhole): [ 600 600 599.5 339.5 ]
        -Camera 1 distortion parameters: [  0.000238406 -0.000314798 -7.39231e-05 -2.7716e-05 0 ]
        -Original image size: [ 1200 , 680 ]
        -Current image size: [ 1200 , 680 ]
        -Sequence FPS: 30
        -RGB-D depth map factor: 6553.5
        -Features per image: 1600
        -ORB scale factor: 1.2
        -ORB number of scales: 8
        -Initial FAST threshold: 20
        -Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
[Gaussian Mapper]CUDA available! Training on GPU.
[Gaussian Mapper]Reading parameters from "./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml"
[ImGuiViewer]Reading parameters from "./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml"

-------
Start processing sequence ...
Images in the sequence: 2000

First KF:0; Map init KF:0
New Map created with 1560 points

Segmentation fault (core dumped)

```

then use gdb
```
root@cf7685d5dec7:/Photo-SLAM# gdb ./bin/replica_rgbd ./.27723 
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
--Type <RET> for more, q to quit, c to continue without paging--c
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/replica_rgbd...
(No debugging symbols found in ./bin/replica_rgbd)

warning: Can't open file /dev/zero (deleted) during file-backed mapping note processing

warning: Can't open file /SYSV00000000 (deleted) during file-backed mapping note processing
[New LWP 27810]
[New LWP 27731]
[New LWP 27723]
[New LWP 27773]
[New LWP 27771]
.......
[New LWP 27866]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `./bin/replica_rgbd ./ORB-SLAM3/Vocabulary/ORBvoc.txt ./cfg/ORB_SLAM3/RGB-D/Repl'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000074f14e5eff92 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
[Current thread is 1 (Thread 0x74f15d7fe000 (LWP 27810))]
(gdb) bt
#0  0x000074f14e5eff92 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#1  0x000074f14e5e896d in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#2  0x000074f14e666bc2 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#3  0x000074f14e654df8 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#4  0x000074f14e639440 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#5  0x000074f14e2232f0 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#6  0x000074f14e224e29 in ?? () from /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
#7  0x000074f302b61ea8 in ImGui_ImplOpenGL3_RenderDrawData(ImDrawData*) () from /Photo-SLAM/lib/libimgui.so
#8  0x000074f31d1e8ae0 in ImGuiViewer::run() () from /Photo-SLAM/lib/libgaussian_viewer.so
#9  0x000074f2ab4b0253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x000074f2ab23fac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#11 0x000074f2ab2d1850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

I don't understand what happened.