HuajianUP/Photo-SLAM #13: > Hi Jiyu,

xiangw369, 2024-06-12:
> Hi Jiyu,
> 
> We notice that the system is not compatible with the latest Pytorch. Please use the tested version.
> 
> ```
> wget https://download.pytorch.org/libtorch/cu118/libtorch-cxx11-abi-shared-with-deps-2.0.1%2Bcu118.zip -O libtorch-cu118.zip
> unzip libtorch-cu118.zip && rm libtorch-cu118.zip
> 
> # in CMakeLists.txt
> set(Torch_DIR ./libtorch/share/cmake/Torch)
> ```

Thank you so much about your kind reply! 
I've noticed this issue and I compile your code successfully in docker with the dockerfile provided by https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703.
However, when I try to run it, it seems another error occurred. So confusing it is that seems nobody have met this before. Could you please check for it and provide some suggestion?
Thanks a lot!
[image]

_Originally posted by @Jiyu-Yao in https://github.com/HuajianUP/Photo-SLAM/issues/12#issuecomment-2160430352_

closed 2024-06-12 by xiangw369

kunyuwan, 2024-06-13:
Please tell me how to solve it, I also encountered the same problem

LeihanChen, 2024-08-09:
@xiangw369 @Jiyu-Yao  Any updates on it, encountered the same issue with the build with the same dockerfile.