HuajianUP/Photo-SLAM #15: how to render the image from random viewpoint

vvcatstar, 2024-06-14:
I have successfully run your code. The final results only include the render image and camera outputs; the GS results are not found. If I want to input a random camera pose and render the image from that viewpoint, how should I proceed? Thank you.

HuajianUP, 2024-06-16:
Hi, Thank you for your interest and testing our code.

1) The default path to store the PLY file is `SEQ_RESULTS/xxxxx_shutdown/ply/point_cloud/iteration_xxxxx/point_cloud.ply. You can use any viewers compatible with GS visualization to load the PLY file and check the results. Moreover, you can also refer to our evaluation code.

2) You should ensure the intrinsic model of the iPhone camera is calibrated. As an online system, it will stop optimization once tracking is finished. When we continue to visit new places, the newly added Keyframes and primitives have not been thoroughly optimized. A simple solution is to keep optimizing after processing all frames. In addition, you can adjust the densification parameters for fine-tuning. 

3) We would rectify the input images to undistorted perspective images. Currently, the code is not compatible with fisheye images yet, refer to [camera.h](https://github.com/HuajianUP/Photo-SLAM/blob/f8bfb2f0809c003ccc3fd577dc43c576fcafa4ac/include/camera.h). You can add an OpenCV fisheye model, [cv::fisheye::initUndistortRectifyMap](https://docs.opencv.org/4.x/db/d58/group__calib3d__fisheye.html#ga0d37b45f780b32f63ed19c21aa9fd333). And we will update our code and support it later.

vvcatstar, 2024-06-17:
Thank you for your reply, I will calibrate my camera and then to test.
I want to use /bin/view_result to show the ply result, but error.
The run code is:
```
./bin/view_result ./cfg/gaussian_mapper/Monocular/Replica/replica_mono.yaml ./cfg/ORB_SLAM3/Monocular/Replica/office0.yaml ./results/replica_mono_0/office0/5481_shutdown/ply/point_cloud/iteration_5481/point_cloud.ply
```
The result is:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  [Gaussian Mapper]Unsupported camera model: /home/zyw/data/repo_common/Photo-SLAM/cfg/ORB_SLAM3/Monocular/Replica/office0.yaml
```
The camera.type in office0.yaml is 'Pinhole', which is not 'Pinhole' in gaussian_mapper.cpp. So I change the office0.yaml  'PinHole' to 'Pinhole' and run again. But error again. Here is the log:
```
CUDA available! Training on GPU.
[Gaussian Mapper]CUDA available! Training on GPU.
[Gaussian Mapper]Reading parameters from "./cfg/gaussian_mapper/Monocular/Replica/replica_mono.yaml"
        [ply_header] Type: binary
        [ply_header] element: vertex (80535)
        [ply_header]    property: x (type=float)
        [ply_header]    property: y (type=float)
        [ply_header]    property: z (type=float)
        [ply_header]    property: nx (type=float)
        [ply_header]    property: ny (type=float)
        [ply_header]    property: nz (type=float)
        [ply_header]    property: f_dc_0 (type=float)
        [ply_header]    property: f_dc_1 (type=float)
        [ply_header]    property: f_dc_2 (type=float)
        [ply_header]    property: f_rest_0 (type=float)
        [ply_header]    property: f_rest_1 (type=float)
        [ply_header]    property: f_rest_2 (type=float)
        [ply_header]    property: f_rest_3 (type=float)
        [ply_header]    property: f_rest_4 (type=float)
        [ply_header]    property: f_rest_5 (type=float)
        [ply_header]    property: f_rest_6 (type=float)
        [ply_header]    property: f_rest_7 (type=float)
        [ply_header]    property: f_rest_8 (type=float)
        [ply_header]    property: f_rest_9 (type=float)
        [ply_header]    property: f_rest_10 (type=float)
        [ply_header]    property: f_rest_11 (type=float)
        [ply_header]    property: f_rest_12 (type=float)
        [ply_header]    property: f_rest_13 (type=float)
        [ply_header]    property: f_rest_14 (type=float)
        [ply_header]    property: f_rest_15 (type=float)
        [ply_header]    property: f_rest_16 (type=float)
        [ply_header]    property: f_rest_17 (type=float)
        [ply_header]    property: f_rest_18 (type=float)
        [ply_header]    property: f_rest_19 (type=float)
        [ply_header]    property: f_rest_20 (type=float)
        [ply_header]    property: f_rest_21 (type=float)
        [ply_header]    property: f_rest_22 (type=float)
        [ply_header]    property: f_rest_23 (type=float)
        [ply_header]    property: f_rest_24 (type=float)
        [ply_header]    property: f_rest_25 (type=float)
        [ply_header]    property: f_rest_26 (type=float)
        [ply_header]    property: f_rest_27 (type=float)
        [ply_header]    property: f_rest_28 (type=float)
        [ply_header]    property: f_rest_29 (type=float)
        [ply_header]    property: f_rest_30 (type=float)
        [ply_header]    property: f_rest_31 (type=float)
        [ply_header]    property: f_rest_32 (type=float)
        [ply_header]    property: f_rest_33 (type=float)
        [ply_header]    property: f_rest_34 (type=float)
        [ply_header]    property: f_rest_35 (type=float)
        [ply_header]    property: f_rest_36 (type=float)
        [ply_header]    property: f_rest_37 (type=float)
        [ply_header]    property: f_rest_38 (type=float)
        [ply_header]    property: f_rest_39 (type=float)
        [ply_header]    property: f_rest_40 (type=float)
        [ply_header]    property: f_rest_41 (type=float)
        [ply_header]    property: f_rest_42 (type=float)
        [ply_header]    property: f_rest_43 (type=float)
        [ply_header]    property: f_rest_44 (type=float)
        [ply_header]    property: opacity (type=float)
        [ply_header]    property: scale_0 (type=float)
        [ply_header]    property: scale_1 (type=float)
        [ply_header]    property: scale_2 (type=float)
        [ply_header]    property: rot_0 (type=float)
        [ply_header]    property: rot_1 (type=float)
        [ply_header]    property: rot_2 (type=float)
        [ply_header]    property: rot_3 (type=float)
        Read 80535 total xyz 
        Read 80535 total f_dc 
        Read 80535 total f_rest 
        Read 80535 total opacity 
        Read 80535 total scales 
        Read 80535 total rot 
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.7.0) /home/zyw/data/environment/opencv470/opencv-4.7.0/modules/imgproc/src/imgwarp.cpp:1703: error: (-215:Assertion failed) !_map1.empty() in function 'remap'

[1]    2999218 abort (core dumped)  ./bin/view_result ./cfg/gaussian_mapper/Monocular/Replica/replica_mono.yaml 
```

HuajianUP, 2024-06-18:
1) For evaluation, please refer to [Photo-SLAM-eval](https://github.com/HuajianUP/Photo-SLAM-eval)

2) Simply modifying the .yaml file won't work for fisheye images. We need to change the undistortion function in [camera.h](https://github.com/HuajianUP/Photo-SLAM/blob/f8bfb2f0809c003ccc3fd577dc43c576fcafa4ac/include/camera.h) to support the KB camera model.

closed 2024-06-25 by vvcatstar

XU0510, 2025-02-21:
> Thank you for your reply, I will calibrate my camera and then to test. I want to use /bin/view_result to show the ply result, but error. The run code is:
>
> ```
> ./bin/view_result [...]

I found the reason why the program wasn't running successfully. In the gaussian_mapper.cpp file under Photo-SLAM-main/src, there was an error in reading the input camera parameters. The parameter names were incorrect. Originally, it was written as camera.width_ = camera_file["Camera.w"].operator int();, and I changed it to camera.width_ = camera_file["Camera.width"].operator int(); std::cout << "Input camera image size: " << camera.width_ << std::endl; camera.height_ = camera_file["Camera.height"].operator int();. After making this change, the program runs normally. However, I still don't understand why the visualization interface is black and showing nothing.