HuajianUP/Photo-SLAM #17: Building PhotoSLAM issue

shashankyld, 2024-06-18:
Hello. I have issue while building PhotoSLAM. Everything up to PhotoSLAM (i.e. DBoW, g2o, Sophus, ORB-SLAM3) builds correctly, but when it comes to PhotoSLAM part I get this error:
```
Building Photo-SLAM ...
-- The CXX compiler identification is GNU 8.4.0
-- The CUDA compiler identification is unknown
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting CUDA compiler ABI info
CMake Error in /home/shashank/Documents/UniBonn/Sem4/ThesisPrep/Photo-SLAM/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  CUDA_ARCHITECTURES is empty for target "cmTC_33549".

CMake Error in /home/shashank/Documents/UniBonn/Sem4/ThesisPrep/Photo-SLAM/build/CMakeFiles/CMakeTmp/CMakeLists.txt:
  CUDA_ARCHITECTURES is empty for target "cmTC_33549".

CMake Error at /opt/cmake-3.22.0/share/cmake-3.22/Modules/CMakeDetermineCompilerABI.cmake:49 (try_compile):
  Failed to generate test project build system.
Call Stack (most recent call first):
  /opt/cmake-3.22.0/share/cmake-3.22/Modules/CMakeTestCUDACompiler.cmake:19 (CMAKE_DETERMINE_COMPILER_ABI)
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/shashank/Documents/UniBonn/Sem4/ThesisPrep/Photo-SLAM/build/CMakeFiles/CMakeOutput.log".
See also "/home/shashank/Documents/UniBonn/Sem4/ThesisPrep/Photo-SLAM/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
```

My PC is:
Ubuntu 20.04
nvcc version: 11.8 (I installed it in conda environment)
gcc version: 13.2
cmake version: 3.22

Please if you know what is the issue. I am using **conda environment** just to keep in mind (nvcc is there and other cuda things)

HuajianUP, 2024-06-18:
Hi, what kind of GPU are you using? And is the Pytorch version compatible?

666zachary, 2024-07-02:
I also encountered the same problem, have you solved it?