HuajianUP/Photo-SLAM #20: ZED 2 Camera?

ecervera, 2024-07-04:
Congratulations for your amazing work.
I read in your paper that you have used a ZED 2 stereo camera to collect outdoor scenes.
But I couldn't find the code for ZED 2 in the repository, only for the RealSense.
Could the ZED code be available?
Many thanks!

HuajianUP, 2024-07-08:
Hi, we don't have live demos with ZED2. We used ZED SDK to collect frames and then run the experiments. You can refer to the EuRoC stereo example. If you want to access our stereo data, I can share it with you via cloud storage.

closed 2024-07-08 by ecervera

ecervera, 2024-07-12:
Dear Huajian,

I would appreciate it if you could share the stereo data via cloud storage.

Best regards,
Enric

Missatge de Huajian ***@***.***> del dia dl., 8 de jul. 2024
a les 4:34: