HuajianUP/Photo-SLAM #22: Export the result of Photo-SLAM Examples with Real Cameras

Tanfuchy, 2024-07-21:
I can build the map in real time by using realsense provided by you, but I don't know how to save this result, such as exporting it as ply file. Thank you.

liquorleaf, 2024-09-08:
If the imgui viewer is available, just close the viewer and the results are automatically saved. Otherwise you might need to manually add some save points by changing `GaussianMapper::savePly` to a public member and calling it in the `while (!pSLAM->isShutDown())` main loop.

respect-jz, 2024-10-07:
> 如果 imgui 查看器可用，只需关闭查看器，结果就会自动保存。否则，您可能需要通过更改`GaussianMapper::savePly`为公共成员并在`while (!pSLAM->isShutDown())`主循环中调用它来手动添加一些保存点。

你好，我应该是正确使用了您提供的D455相机使用到该项目，但是在结果文件中发现只有used_times文件夹、没有ply文件。我应该怎么保存ply文件