HuajianUP/Photo-SLAM #23: D435

SSU-K, 2024-07-26:
Hi,

Thanks for your work.

I wanna try this work with D435 RGB-D camera.
Is there any way for this?

ZhangHX-2514, 2024-08-08:
Hi, how's your work going? I'm working on something similar recently. Do you have any advice?

harshal-14, 2024-08-18:
Hi,
Were you able to test this on Realsense D435i camera?

ZhangHX-2514, 2024-08-19:
> Hi, Were you able to test this on Realsense D435i camera?

The results we have obtained so far are not very satisfactory. How about yours?

Adithya1911, 2025-04-09:
Hi,
I want to work on the Realsense 435D camera, so i have written the yaml file for the orbslam and unable to written any thing in the config file for the Gaussian Mapper and getting this error even for the 455D also 
./realsense_d455.sh 
Sensor supports the following options:

  0: Backlight Compensation is not supported
  1: Brightness is not supported
  2: Contrast is not supported
  3: Exposure
       Description   : Depth Exposure (usec)
       Current Value : 33000
  4: Gain
       Description   : UVC image gain
       Current Value : 16
  5: Gamma is not supported
  6: Hue is not supported
  7: Saturation is not supported
  8: Sharpness is not supported
  9: White Balance is not supported
  10: Enable Auto Exposure
       Description   : Enable Auto Exposure
       Current Value : 1
  11: Enable Auto White Balance is not supported
  12: Visual Preset
       Description   : Advanced-Mode Preset
       Current Value : 0
  13: Laser Power
       Description   : Manual laser power in mw. applicable only when laser power mode is set to Manual
       Current Value : 150
  14: Accuracy is not supported
  15: Motion Range is not supported
  16: Filter Option is not supported
  17: Confidence Threshold is not supported
  18: Emitter Enabled
       Description   : Emitter select, 0-disable all emitters, 1-enable laser, 2-enable laser auto (opt), 3-enable LED (opt)
       Current Value : 1
  19: Frames Queue Size
       Description   : Max number of frames you can hold at a given time. Increasing this number will reduce frame drops but increase latency, and vice versa
       Current Value : 16
  20: Total Frame Drops is not supported
  21: Auto Exposure Mode is not supported
  22: Power Line Frequency is not supported
  23: Asic Temperature is not supported
  24: Error Polling Enabled
       Description   : Enable / disable polling of camera internal errors
       Current Value : 1
  25: Projector Temperature is not supported
  26: Output Trigger Enabled
       Description   : Generate trigger from the camera to external device once per frame
       Current Value : 0
  27: Motion Module Temperature is not supported
  28: Depth Units
       Description   : Number of meters represented by a single depth unit
       Current Value : 0.001
  29: Enable Motion Correction is not supported
  30: Auto Exposure Priority is not supported
  31: Color Scheme is not supported
  32: Histogram Equalization Enabled is not supported
  33: Min Distance is not supported
  34: Max Distance is not supported
  35: Texture Source is not supported
  36: Filter Magnitude is not supported
  37: Filter Smooth Alpha is not supported
  38: Filter Smooth Delta is not supported
  39: Holes Fill is not supported
  40: Stereo Baseline
       Description   : Distance in mm between the stereo imagers
       Current Value : 95.2081
  41: Auto Exposure Converge Step is not supported
  42: Inter Cam Sync Mode
[... 40 lines cut ...]
  73: Trigger Camera Accuracy Health is not supported
  74: Reset Camera Accuracy Health is not supported
  75: Host Performance is not supported
  76: Hdr Enabled
       Description   : HDR Option
       Current Value : 0
  77: Sequence Name
       Description   : HDR Option
       Current Value : 0
  78: Sequence Size
       Description   : HDR Option
       Current Value : 2
  79: Sequence Id
       Description   : HDR Option
       Current Value : 0
  80: Humidity Temperature is not supported
  81: Enable Max Usable Range is not supported
  82: Alternate IR is not supported
  83: Noise Estimation is not supported
  84: Enable IR Reflectivity is not supported
  85: Auto Exposure Limit
       Description   : Exposure limit is in microseconds. Default is 0 which means full exposure range. If the requested exposure limit is greater than frame time, it will be set to frame time at runtime. Setting will not take effect until next streaming session.
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
./realsense_d455.sh: line 7: 45571 Aborted                 (core dumped) ../bin/realsense_rgbd ../ORB-SLAM3/Vocabulary/ORBvoc.txt ../cfg/ORB_SLAM3/RGB-D/RealCamera/realsense_d455_rgbd.yaml ../cfg/gaussian_mapper/RGB-D/RealCamera/realsense_rgbd.yaml ../results/realsense_d455_rgbd