HuajianUP/Photo-SLAM #25: iPad Pro and the NeRFCapture app

sjdingcn, 2024-08-02:
Hi,
Thank you for the great work.
May I ask if it is possible to use iPad Pro and the NeRFCapture app with this project?
How should I change the config files and what else should I change?

Thank you
Best

liquorleaf, 2024-09-08:
You might need to modify the whole rendering module, since the rendering pipeline of NeRF (volume-based) and 3D Gaussian Splatting (point-based) is largely different.

sjdingcn, 2024-09-09:
Thank you for your reply. I have one more question.
If I need to freeze some Gaussians during training, how do you think I should modify the code?

liquorleaf, 2024-09-10:
Properties of Gaussians are managed by SOAs during training. Please refer to `GaussianModel::densifyAndClone` and `GaussianModel::densifyAndSplit` to see how to make a `selected_pts_mask`, which indicates the points you want to optimize.

sjdingcn, 2024-09-23:
Thanks. This is very helpful. 
Another question, how to render the trajectory like Figure 13 in the paper?