HuajianUP/Photo-SLAM #30: Interface display issue

dongxiaozi, 2024-09-02:
Hello, thank you for your excellent work. I successfully compiled the Ubuntu 18.04 system according to the versions of other packages you provided, but there was a problem with the display interface during the final run. The current render frame and the overall field of view in the middle were flashing ellipses of various color or black screens, which did not display the correct results. However, there were no errors in the terminal. What is the reason for this situation? Is there any solution? Thank you. 
[image]

dongxiaozi, 2024-09-02:
[image]

liquorleaf, 2024-09-08:
1. Are the original 3DGS and SIBR_viewer working properly on your computer?
2. Are the image files rendered after shutdown correct?
If so, there may be some problem with the glfw-based ImGUI.

mirpouya, 2025-01-16:
> [image]

how could you bring the visualization I face this issue:
I did:
(base) pouya@mir:~/Desktop$ xhost +local:docker 
non-network local connections being added to access control list 

(base) pouya@mir:~/Desktop$ sudo docker run -it --gpus all \
    -e DISPLAY=$DISPLAY \
    -v /tmp/.X11-unix:/tmp/.X11-unix \
    photo-slam:work-in-progress

root@85bfd960eeec:/Photo-SLAM# ./bin/replica_rgbd \
    ./ORB-SLAM3/Vocabulary/ORBvoc.txt \
    ./cfg/ORB_SLAM3/RGB-D/Replica/office0.yaml \
    ./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml \
    /Photo-SLAM/data/Replica/office0 \
    /Photo-SLAM/results

and I faced:

root@85bfd960eeec:/Photo-SLAM# ./bin/replica_rgbd \
    ./ORB-SLAM3/Vocabulary/ORBvoc.txt \
    ./cfg/ORB_SLAM3/RGB-D/Replica/office0.yaml \
    ./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml \
    /Photo-SLAM/data/Replica/office0 \
    /Photo-SLAM/results
CUDA available! Training on GPU.

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D
Loading settings from ./cfg/ORB_SLAM3/RGB-D/Replica/office0.yaml
	-Loaded camera 1
Camera.newHeight optional parameter does not exist...
Camera.newWidth optional parameter does not exist...
	-Loaded image info
	-Loaded RGB-D calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 600 600 599.5 339.5 ]
	-Camera 1 distortion parameters: [  0.000238406 -0.000314798 -7.39231e-05 -2.7716e-05 0 ]
	-Original image size: [ 1200 , 680 ]
	-Current image size: [ 1200 , 680 ]
	-Sequence FPS: 30
	-RGB-D depth map factor: 6553.5
	-Features per image: 1600
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
[Gaussian Mapper]CUDA available! Training on GPU.
[Gaussian Mapper]Reading parameters from "./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml"
[ImGuiViewer]Reading parameters from "./cfg/gaussian_mapper/RGB-D/Replica/replica_rgbd.yaml"

-------
Start processing sequence ...
Images in the sequence: 2000

First KF:0; Map init KF:0
New Map created with 1560 points
Segmentation fault (core dumped)
root@85bfd960eeec:/Photo-SLAM# echo $DISPLAY
:1
root@85bfd960eeec:/Photo-SLAM# 

every time this happens and I can't track the issue 

can you please help me