HuajianUP/Photo-SLAM #32: How can I build a yaml ？

Suhao07, 2024-09-06:
If I want to run on another dataset, how can I obtain the YAML file for Gaussian Mapper?

liquorleaf, 2024-09-08:
You will need a YAML for ORB-SLAM3, e.g. cfg/ORB_SLAM3/Monocular/Replica/office0.yaml, and another YAML for Gaussian Mapper, e.g. cfg/gaussian_mapper/Monocular/Replica/office0.yaml.

Then you will have to adjust the parameters in both YAMLs:
For ORB-SLAM3,  adjust the parameters according to your type of sensor (there are also templates in cfg/ORB_SLAM3/Stereo and cfg.ORB_SLAM3/RGB-D) and image resolution.
For Gaus. Mapper, adjust the parameters started with Monocular./Stereo./RGBD. according to your type of sensor.

Suhao07, 2024-09-12:
> You will need a YAML for ORB-SLAM3, e.g. cfg/ORB_SLAM3/Monocular/Replica/office0.yaml, and another YAML for Gaussian Mapper, e.g. cfg/gaussian_mapper/Monocular/Replica/office0.yaml.
>
> Then you will [...]

> You will need a YAML for ORB-SLAM3, e.g. cfg/ORB_SLAM3/Monocular/Replica/office0.yaml, and another YAML for Gaussian Mapper, e.g. cfg/gaussian_mapper/Monocular/Replica/office0.yaml.
>
> Then you will [...]

liquorleaf, 2024-09-12:
This is probably not a YAML problem. Please see if #5 #12 #15 #33 solves it.

Suhao07, 2024-09-12:
> This is probably not a YAML problem. Please see if #5 #12 #15 #33 solves it.

I tried these methods but it didn't work.Have you test it on  tum_freiburg3_walking_xyz before?When I tested on other datsets like tum_freiburg1_desk, I didn't encounter this issue.Or I mean  have you tested on dynamic scene before? Thanks for your reply:)

liquorleaf, 2024-09-12:
Please see [the ORB-SLAM3 association files for TUM](https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/master/Examples/RGB-D/associations)