HuajianUP/Photo-SLAM #33: Segmentation fault (core dumped)

Suhao07, 2024-09-07:
[image]

liquorleaf, 2024-09-08:
Might be a ORB-SLAM3 problem. Please see whether [ORB_SLAM3 issue 443](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/443) resolves it.

Santasu28, 2024-11-07:
Same as you，have you solved this issue？

Suhao07, 2024-11-11:
> Same as you，have you solved this issue？

I've tried lots of solutions but all of them didn't work at all:(
After that I doubt that  the dataset is incomplete or broken so I download TUM dataset on their official website ,and I didn't encounter the same problem any more.
btw the dataset I used is dynamic scene,when I ran the code on static scenes it works fine.So u can try the solutions of ORB_SLAM3,and check your dataset.Hope it could help u :D

Santasu28, 2024-11-11:
> I've tried lots of solutions but all of them didn't work at all:( After that I doubt that the dataset is incomplete or broken so I download TUM dataset on their official website ,and I didn't [...]

Thank you a lot ! Kind of you!!

XYJ88888, 2024-11-20:
I also had this problem, opening viewer would cause segmentation fault, but when I use no_viewer it works fine

ERGOWHO, 2024-12-25:
same problem solved by use "no_viewer"

mirpouya, 2025-01-16:
same problem here, but I want to see the real-time performance and mapping of this model, without visualization it's like the biggest part is missing

> I also had this problem, opening viewer would cause segmentation fault, but when I use no_viewer it works fine

mirpouya, 2025-01-16:
there's good guideline in #5 
I solved the issue with this docker file:

FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    libssl-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
    wget https://download.pytorch.org/libtorch/nightly/cu118/libtorch-cxx11-abi-shared-with-deps-2.4.0.dev20240425%2Bcu118.zip -O libtorch-cu118.zip && \
    unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM

everything is the same as https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703
but the libtorch version is 2.4.0

then you must modify all of:
     c10::guts::to_string(param.unsafeGetTensorImpl());
to:
     param.unsafeGetTensorImpl();
which is mentioned in #5 

and you will see the visualization

yuanzou608, 2025-05-12:
Hi,
    I still get error the Segmentation fault  (core dumped) with different Dockerfile  or dataset, but with no_viewer works well. Do you have another solution?

This is error information

Start processing sequence ...
Images in the sequence: 3000

First KF:0; Map init KF:0
New Map created with 1206 points
./my_rgbd.sh: line 3:   916 Segmentation fault      (core dumped) ../bin/tum_rgbd ../ORB-SLAM3/Vocabulary/ORBvoc.txt ../cfg/ORB_SLAM3/RGB-D/airsim.yaml ../cfg/gaussian_mapper/RGB-D/airsim.yaml /home/yuan/dataset/my_data/RGBD18fps /home/yuan/dataset/my_data/RGBD18fps/associations.txt ../results/my_resluts

Suhao07, 2025-05-14:
Nope,sorry that I don't have other solutions发自我的手机-------- 原始邮件 --------发件人： yuanzou608 ***@***.***>日期： 2025年5月12日周一 晚上7:03收件人： HuajianUP/Photo-SLAM ***@***.***>抄送： rickyman ***@***.***>, Author ***@***.***>主    题： Re: [HuajianUP/Photo-SLAM] Segmentation fault (core dumped) (Issue #33)yuanzou608 left a comment (HuajianUP/Photo-SLAM#33)
Hi,
I still get error the Segmentation fault  (core dumped) with different Dockerfile  or dataset, but with no_viewer works well. Do you have another solution?
This is error information
Start processing sequence ...
Images in the sequence: 3000
First KF:0; Map init KF:0
New Map created with 1206 points
./my_rgbd.sh: line 3:   916 Segmentation fault      (core dumped) ../bin/tum_rgbd ../ORB-SLAM3/Vocabulary/ORBvoc.txt ../cfg/ORB_SLAM3/RGB-D/airsim.yaml ../cfg/gaussian_mapper/RGB-D/airsim.yaml /home/yuan/dataset/my_data/RGBD18fps /home/yuan/dataset/my_data/RGBD18fps/associations.txt ../results/my_resluts